unreality/FujiHeatPump #2: Error EE: 1C (Outdoor unit computer communication error)

chemuki205, 2021-11-02:
Hello

First of all, thank you for this work, it is something that I have been looking for for a long time to be able to integrate one aircon in a home automation installation.

I have followed the steps to do it on an esp32, but I have found a problem that will surely be related to the connection of the slave.

Could you explain in detail how you have connected the system to the air conditioning?
Following the instructions, the aircon shows an EE: 1C error (Outdoor unit computer communication error) and the main control is disconnected by turning off the aircon.

Is there a connection / power-up sequence for both the aircon and the esp32 so that it is detected as a slave without problems?

Thanks for the help

unreality, 2021-12-31:
Sorry I didnt see this before now. I'll attempt to explain the boot up sequence to the best of my memory, I havent looked at this for a while :)

When the AC powers up it continuously sends out frames on the 3-wire bus looking for a primary controller. When a primary controller is finished booting it will wait (a few minutes from my testing, but i dont know the exact time) for frames from the indoor unit.

Once it gets a frame, it announces itself to the indoor unit by sending a login frame, and the indoor unit will respond by sending status frames.

When it receives its first status frame, the primary controller will search for a secondary controller by sending a single frame to the secondary controller address. If the secondary controller successfully forwards that frame back to the unit the primary controller will continue to send frames to the secondary controller. Otherwise it considers itself the only controller and never attempts to send frames to the secondary again.

This means that to add a secondary controller, you need to power cycle the entire system (and also set the primary controller to forget its settings by toggling the dip switch on the back)

chemuki205, 2022-02-15:
Thanks for the answer, unfortunately I have left this parked due to lack of time, as soon as I can get back to it I will try it and comment on the result.

Thanks again

krahabb, 2022-08-11:
Hello @unreality,
Thank you for your code. I'm too getting mad trying to get it working as a secondary controller.
My master shows a couple of 9C during boot and then nothing. Just the master (wall mounted controller) works and the esp sits there without being able to 'announce' itself or interact with the bus.
I've done a lot of testing and I can see the master querying out for the secondary (around 4 sec after boot) on the bus. The secondary (my esp) then sends out its status frame but the primary looks like it is not recognizing it and stops broadcasting frames to the slave.
I've tried to modify the response delay in code guessing it was a timing problem on the bus but nothing.

In a previous post you suggest to reset the primary controller settings by a dip switch but I can't find any source of information on this. I see the DIP1-2 is used to set the controller unit as master or slave but I can't really understand the meaning of the other switches (fujitsu docs mainly say they're factory set and to not touch)

unreality, 2022-08-12:
@krahabb  I believe its the same DIP switch - PIN 1 sets the primary/secondary, and pin 2 turns on/off remembering information (such as the clock time) when the system loses power.