unreality/FujiHeatPump #3: in secondary ctrl mode, cannot set ECO

rabbit-aaron, 2022-01-03:
Hey! Great work buddy.

I'm integrating your work with ESPHome and connecting it with Home Assistant (will publish when it's more stable).

I have my ESP32 running as the secondary controller.
It seems to work really well! However, I can't set the Economy mode.
I tried `heatPump.setEconomyMode(true)` as well as `heatPump.setEconomyMode(static_cast<byte>(1))`;

I was able to get the correct ECO status from the main controller though.

unreality, 2022-01-17:
Unfortunately my units dont have an economy mode so I couldnt test that mode. You may need to enable the debug logging and inspect the frames manually to make sure its being set correctly.

If you manage to fix it, a merge request would be great :)

chemuki205, 2022-02-15:
That is what I intend, to integrate 2 AA devices in Home Assistant, I would be very grateful if you can publish it when it is stable

Thanks

rabbit-aaron, 2022-02-15:
I have the code running fine but when it loses WiFi connection, it will miss some frames and cause the master controller to fail and enter an error state, when that happens I have to power cycle the entire AC.

I plan on limiting ESPHome to only 1 core and dedicating the other core to handling the serial communication

unreality, 2022-02-16:
For the FujiHK i made the serial comms pinned to core 1, https://github.com/unreality/FujiHK/blob/c646f70976909e1f1ec22af493a9c4b84b0458d0/src/fuji-hk.cpp#L560 may be useful if you want some sample code to look at...

rabbit-aaron, 2022-02-16:
Nice one, thanks!

I'll have to work it out with ESPHome though, they run RTOS I believe and some tasks are using that second core, I'll probably jump on their discord and find out more.

unreality, 2022-02-16:
If you really want to go crazy, you could use the ULP to handle the serial comm with the HeatPump, it should have enough power to do it, and it can control a serial port. It was on my list of things to try but the extra core seems to work well enough for now.

rabbit-aaron, 2022-02-16:
wow if that works we don't need to waste that extra core, +1 for that feature haha

I don't deal with microprocessors a lot, I'm a web developer, a lot of these is very confusing to me :(

rabbit-aaron, 2022-04-02:
> That is what I intend, to integrate 2 AA devices in Home Assistant, I would be very grateful if you can publish it when it is stable
> 
> Thanks

buddy i got mine with dual-core working all thanks to @unreality 's sample code.
If you are interested see here: https://github.com/rabbit-aaron/esphome-fujitsu/tree/dual-core

unreality, 2023-01-23:
@rabbit-aaron i have just committed a change that should make eco mode work, try it out and see if you can toggle it. The update code wasnt actually sending changes to eco mode

rabbit-aaron, 2023-01-24:
Yo thank you very much, I'll update the people in this post too.

Huge thanks!

closed 2023-11-08 by unreality