unreality/FujiHeatPump #4: HW documentation

pacmyc, 2022-08-30:
Hi.
I'd love to try to use your library.
But I can't find any info about my unit, model Fuji RSG12LMBC; whether it has the three wire interface or RX/TX lines, and where these are situated?
Any help / details would be deeply appreciated..

unreality, 2022-08-31:
Im afraid i cant find any results for that model number, are you certain its accurate?

Normally searching for the model will yield a service manual of some kind, which you can have a look at to determine the remote type. Alternatively you can look up the model of your remote control units and see if they are the 3-wire type.

pacmyc, 2022-08-31:
Sorry, I mixed up the last two letters, it should be RSG12LMCB.
I only have a IR remote for the device. The operator manual however states that it is possible to connect a wired remote for it as well, but doesn't mention any model or name of the remote.

The closest match of service manual I could find is this https://www.manualslib.com/download/1204533/Fuji-Electric-Rsg12leca.html .
On page 6 it displays a "communication pcb assembly" which is marked as "option". It has input and output terminals for wired remote control.

Any ideas how to check if I have this board? I guess the terminals should be easy access in order to install a wired remote.. Where should I look for the terminals?

abhinav626, 2024-04-05:
If your thing has an IR then its easier to send an IR signal from any esp board ( You can Tasmota-IR),and use a temperature sensor  for feedback...