unreality/FujiHeatPump #5: Zone control commands

bezmi, 2022-11-16:
Hi, I'd like to interface with the my system's UTY-RVNYN controller, which allows for toggling zones provided a zone control interface is attached (see page 8 of the attached pdf). Do you have any recommendations for figuring out the commands needed? If use this as a secondary controller and set `debugPrint` to true, will it be enough to capture the required data?

[DR_OP002EF_06-UTY-RVNYN_UTY-CDPXZC-D_UTY-TFNXZ3_FGLair-1.pdf](https://github.com/unreality/FujiHeatPump/files/10023532/DR_OP002EF_06-UTY-RVNYN_UTY-CDPXZC-D_UTY-TFNXZ3_FGLair-1.pdf)

unreality, 2022-11-16:
Hi @bezmi 

That sounds like a good start, and it should provide all the information. As far as i know the frames are only ever 8 bytes long. Be aware that many of the bytes are bitfields, so it may be useful to change the debug print function to also output the bytes in binary so you can see which bits are flipped.

closed 2022-11-24 by bezmi

j-marz, 2023-10-27:
@bezmi did you end up figuring out the commands for zone control?

bezmi, 2023-12-13:
> @bezmi did you end up figuring out the commands for zone control?

Unfortunately not, we moved and the new place has a different system.