unreality/FujiHeatPump #6: question can i use this for controlling my ac with esp32 that not have wifi units inside.

pbvdven, 2023-01-27:
I have 4 Fujitsu general aircon but only 2 of them have addon wifi units inside is it possible to fit the other 2 units with this project on esp32 and add then to home assistant? 

i have ac unit ASHG14KMCC.

Tanks

unreality, 2023-01-28:
should be possible, you'll need to write the code to interact with home-assistant though

pbvdven, 2023-01-28:
> should be possible, you'll need to write the code to interact with home-assistant though

Thanks

closed 2023-01-28 by pbvdven

pbvdven, 2023-01-28:
> should be possible, you'll need to write the code to interact with home-assistant though

My unit does not have an the optional communions kit needed for the 3 wire remote. i only have ir-remotecontrol but it has the connection for the wifi module but its 4 wires instead of 3 any idea how to make that work? [image]

unreality, 2023-01-29:
Have a look at https://github.com/jaroslawprzybylowicz/fuji-iot 

It includes a PCB layout and BOM that hooks directly into the internal unit. I dont think it needs the communications PCB, but i could be mistaken.