unreality/FujiHeatPump #7: Sending temperature to heat pump

lukepighetti, 2023-03-14:
Is there any way to inform the heat pump of the actual room temperature? This would allow a high precision external thermocouple to be installed in a central area in the room improving the accuracy of the temperature data available to the heat pump's internal control unit.

unreality, 2023-03-15:
If you want a dedicated temperature sensor, i'm unsure whether the protocol supports that.

You could create a dumb controller using this library and change `waitForFrame()` so that it sends the temperature from your sensor instead of copying the one it sees from the other controller. Or create a controller that accepts updates from other temperature sensors to relay back to the heatpump.

lukepighetti, 2023-03-15:
> Or create a controller that accepts updates from other temperature sensors to relay back to the heatpump.

I think that's my core question: how do we inform the heat pump of an external temperature sensor value?

unreality, 2023-03-16:
> I think that's my core question: how do we inform the heat pump of an external temperature sensor value?

The frames sent by the controllers contains a 'controllerTemperature', that the only way I know at the moment

nahall, 2023-09-07:
Did you ever get this to work? This is my same goal, to be able to tell the minisplit what the room temperature is from an wireless temperature sensor away from the unit, to reduce the amount of cycling the minisplit does.

unreality, 2023-09-07:
I have not worked on this since i dont really have a need, however it may be an open question as to whether the unit actually uses the temperatures received by the remotes. Im told there may be a sensor in the return duct that the units will use by default instead. I remember reading somewhere there may be some hardware dip switches or settings that make it use the remote temperature settings instead? But im afraid a cant remember where i read that at the moment :/

tevaadarelsol, 2023-12-03:
Also interested on it. Any news?

robfox92, 2025-04-02:
A bit of thread necromancy, but in the installation manual for UTY-RVNYN it talks about this a little. You can get to the "function setting" menu by:
1. Starting at the "monitor" screen
2. Pressing "menu" twice
3. Holding both screen switch buttons (left + right) for 5s
4. Selecting "Function setting"

Function 42 and 48 control where it gets the temperature from. 

- Function 42 allows use of the remote controller
- Function 48 disables use of the indoor unit (I believe this is the return duct?)

Source of temperature measurements:

.                 | 42 = False | 42 = True |
---------------|----------------|---------------|
48 = False  |         IU*     | IU+RC              |
48 = True |          ???       |  RC

\* Default behaviour

I'm unsure if 42 = False and 48 = True is valid.

jpa73, 2025-11-01:
Not an issue as such but super interested in how to get this working.

My system has a ducted main unit in the roof and 2 identical wired controllers (primary & Secondary).  Function 42 is set to enabled and what that appears to do is allow the sensor from the secondary unit to be selected on a dropdown that appears on the zones page of the primary controller. That dropdown allows me to select 'indoor unit'  'primary' or  'secondary' temperature sensor. We normally use this in the timer functions when we want the temperature from the secondary unit to determine the actions of the system.  

I removed the secondary controller and installed my esp32 running oxadam fork with zone control which acts as the interface. https://github.com/FujiHeatPump/esphome-fujitsu .  I have got a small touchscreen esp 32 working well as a controller but would like a method to provide the value of a temperature sensor in Home assistant back to the heat pump so that it proxies the sensor of the original secondary controller. This will allow me to maintain my original time settings but potentially have sensors in all zones and determine which we want to act as that Secondary controller sensor.

 With the secondary controller removed the secondary sensor option disappears from the dropdown, I can force it back on through the settings menu to see what it does using the log.  As I switch between the sensor options with all 8 zones on, I get the following frames in the log indicting which bits change. Posting here in case it is useful for someone smarter than me.

Indoor Unit Selected
09:28:55 [I] [frames:322] Zone Update
09:28:55 [I] [zone_decoding:072] 20 A1 58 F7 80 12 56 C1
09:28:55 [I] [zone_decoding:086] Zone group is 0
09:28:55 [I] [frames:322] Zone Update
09:28:55 [I] [zone_decoding:072] 0 A0 50 F7 0 12 56 C1
09:28:55 [I] [zone_decoding:086] Zone group is 0

Primary Selected
09:29:40 [I] [frames:322] Zone Update
09:29:40 [I] [zone_decoding:072] 20 A1 58 F7 81 12 56 C1
09:29:40 [I] [zone_decoding:086] Zone group is 0
09:29:41 [I] [frames:322] Zone Update
09:29:41 [I] [zone_decoding:072] 0 A0 50 F7 1 12 56 C1
09:29:41 [I][zone_decoding:086] Zone group is 0

Secondary 
09:28:16 [I] [frames:322] Zone Update
09:28:16 [I] [zone_decoding:072] 20 A1 58 F7 82 12 56 C1
09:28:16 [I] [zone_decoding:086] Zone group is 0
09:28:17 [I] [frames:322] Zone Update
09:28:17 [I] [zone_decoding:072] 0 A0 50 F7 2 12 56 C1
09:28:17 [I] [zone_decoding:086] Zone group is 0