unreality/FujiHeatPump #8: 2 wire

wrt83, 2023-03-15:
I have a fujitsu ducted heat pump that utilizes the 2 wire protocol.  The line is 12v dc and I suspect/home it's a LIN configuration using the same commands that pulls power on the signal, but I'm not sure how to confirm that.  I have read some documentation that states the internal board has a jumper to switch between 2 and 3 wire thermostats, but I don't have easy access to check and it's not in my models design and tech manual.  My thermostat is the uty-rnruz.  Suggestions on how to check?  Sometime like the design mentioned in this? https://ww1.microchip.com/downloads/en/DeviceDoc/article_ac9_two-wire-lin-networking.pdf Others I'm sure will benefit if we can reverse engineer this one.

sagilo, 2023-07-04:
@wrt83 did you manage to get it working? any other solution for controlling ducted 2 wire heat pump?

wrt83, 2023-07-11:
@sagilo not yet, other projects have had priority.

Saxin, 2025-10-15:
@wrt83 @sagilo Hi guys,

I'm in same situation, I have ducted concealed unit which came with x2 wire wired controller.
My unit has CN65 connector on main PCB which I will use to leech 12Vdc.
My assumption and hope is that the x2 wires LIN protocol is the same as the x3 wire one - the only difference is baud and signal level.

I'm assembling in this way:
CN65 --> 12V + GND --> 5V DCDC --> ESP32
x2 wire bus --> basically same configuration as the author of this repo shows, I have MCP2021-330E IC on hand - LIN connection: one wire to Lbus, second wire to VSS (for reference), all other connection per this repo. Vbb - 12V from CN65, Vss GND from CN65.

Hope to update in few days...

wrt83, 2025-10-22:
That's too bad. I fried the mcu on the main board earlier this year installing a humidifier.  It's not very well protected, so be carful.  The board is really expensive for what it is.

Mcu was a Renesas R5F100PHA, which has LIN connectivity and it looked like the traces went straight to the chip. LIN is the only thing on that chip that could bridge the distance.  Maybe I missing a uart to rs485 chip on the board?  Could be another reasonable option.