unreality/FujiHeatPump #11: Communicating to outdoor unit

kristsm, 2023-08-08:
Hi! I am trying to convert Fujitsu (General) AC  split system (DC Inverter type R410( to air/water heat pump. I have an outdoor unit without indoor unit. Do you thinks it's possible to use this code for communicating with the outdoor unit directly without the indoor unit (turn on/off etc)? Outdoor unit has following terminal connections:, by the wiring schematics they are identified as follows:  (1) N ; (2) L ; (3) Serial; (4) Earth. Does this make it as some kind of 1-wire serial communication interface? Is there any kind of universal protocol for outdoor units or it's proprietary for each manufacturer? 
Thanks for any kind of information in advance!

limones1319, 2023-09-12:
Hello. I'm curious too if this project can replace an indoor unit to control only the outside unit.

@kristsm did you find anything?

unreality, 2023-09-13:
I believe this guy made some progress on decoding the indoor <--> outdoor unit comms:

https://hackaday.io/project/19473-reverse-engineering-a-fujitsu-air-conditioner-unit