unreality/FujiHeatPump #12: Get temperature with decimals

mrbluebrett, 2023-09-05:
Hi, is it possible to get temperature with decimals? Previously I used an Intesishome module and reported temperatures with a resolution of 0.5º. So can we get the temperature with one decimal to get the 0.5º steps with this?

balouken, 2023-09-07:
Older units work without decimals...
The newer units are with decimals.
Normaly you have this problem also with your IR remote controller.

mrbluebrett, 2023-09-07:
It’s the same ac unit. I just replaced the intesis home module by a custom esp32.

balouken, 2023-09-07:
Yes, but you can only adjust temp by 1°C

[image]

mrbluebrett, 2023-09-07:
I ment the ambient temperature. The reading from the actual temperature.

unreality, 2023-09-07:
As far as i know the line protocol doesnt have capacity for 0.5degC resolution -- however there is one unknown bit in the temperature field:

https://github.com/unreality/FujiHeatPump/blob/72e00ec06c672461e258da8dadf207d0e9f133dc/src/FujiHeatPump.cpp#L22

the other possibility is that the Intesishome unit had its own internal sensor?

mrbluebrett, 2023-09-08:
How can i expose this unknown bit into a variable named for example as a variable named unknown bit (with value 0 and 1) so i can log the vale in HA and check if it changes between the change of every degree??