unreality/FujiHeatPump #14: Single and group zone control

j-marz, 2023-10-29:
Firstly, thanks for this library @unreality , just set it up today and it's working great! :muscle: 

I have the `AR-WDD1E` wall controller which has zone control (single and group) support, which I'd like to be able to control via ESP32.

The group feature allows grouping multiple zones and/or temperature sensors. For example, I have 2 zones (upstairs and downstairs) and have created 2 groups so the temperature sensor in the wall controller is linked to downstairs zone and the temperature sensor in the return duct is linked to upstairs zone.

Here's the bytes from debug mode for each zone and group setting:
- downstairs zone on + upstairs zone on: `20 A1 58 3 80 9 0 41`
- downstairs zone off + upstairs zone on: `20 A1 58 2 80 9 0 41`
- downstairs zone on + upstairs zone off: `20 A1 58 1 80 9 0 41`
- day group: `20 A1 58 1 89 9 0 41`
- night group: `20 A1 58 2 90 9 0 41`
- all group: `20 A1 58 3 98 9 0 41`

I've not yet figured out how to add zone control into your library and was hoping you could help or provide some guidance? :pray: 

One thing to note, whenever changing zone or group settings the following error is logged:
`AC ERROR RECV: 20 A1 50 2 90 9 0 41   mSrc: 32 mDst: 33 mType: 1 write: 0 login: 1 unknown: 1 onOff: 0 temp: 16, mode: 1 cP:0 uM:0 cTemp:0 acError:0`

I assume this is because the library isn't aware of zone related settings?

unreality, 2023-11-07:
> I assume this is because the library isn't aware of zone related settings?

I havent looked at this library for a while, but to find out the group/zone commands you may want to change the debug output to print out the individual bytes in binary so you can see what bits are flipped more easily.

You may want to investigate `readBuf[2]` in `decodeFrame()` since there are a number of bits that are unknown. There may be a different 'message type' for group commands etc, and the bitmask used might need to be expanded to capture that etc

Unfortunately I dont have any group or zone capable units (or at least they are not set up for that at the moment) - so im unable to look at this directly :/

LukeScerri, 2023-12-21:
Hi j-marz,

I'm a fellow Aussie, with the same remote as you, looking to control my zone based fujitsu AC. Did you have any luck? I've built the ESPhome project as per [https://community.home-assistant.io/t/fujitsu-ac-heat-pump-integration-via-esphome-esp32/407610](url) but I get an error I think its because, like you, im running it through a zone controller. 
my wall controller: AR-WDD1E 
Outdoor unit: ARTG36LHTA
zone controller: UTYCDPXZC (I think)

Verpz, 2024-08-27:
@LukeScerri @j-marz Did either of you every figure out zone control?

LukeScerri, 2024-08-27:
> @LukeScerri @j-marz Did either of you ever figure out zone control? I'm also in the same situation.

Sorry mate no. I'm not sure how to sniff, read or decode the protocol in order to add those features to the library. 

I'm happy to contribute to the project, but I need some guidance on how to start the process.

j-marz, 2024-08-27:
Change of priorities on my side and didn't go any deeper into reverse engineering the zone control. Hopefully in a few months i'll try to pick up this project again :pray:

Verpz, 2024-08-27:
Thanks both! Appreciate it

0xadam, 2025-03-11:
I have managed to get zone and zone group control working on a fork of https://github.com/FujiHeatPump/esphome-fujitsu (which incorporates this library into an ESPHome component for integration into Home Assistant). I haven't published my code yet as I need to clean it up, but I hope to in the next little bit.

Verpz, 2025-03-11:
You are amazing. I look forward to testing it out

LukeScerri, 2025-03-11:
Thanks @0xadam 
I'm also happy to assist with some testing when the code is ready. My setup is outlined above.

0xadam, 2025-03-13:
Have just pushed up my code to https://github.com/0xadam/esphome-fujitsu. The version of the library used there should be in sync  with whats here on master i.e. fixes to economy support and reported temperatures above 32c.

Let me know how you go :)

0xadam, 2025-03-21:
Someone else has confirmed it works for them through home assistant, so hopefully works for you both too @Verpz @LukeScerri

LukeScerri, 2025-03-21:
Actually that person was me mate :) thanks again.

Verpz, 2025-03-25:
Hey @0xadam I've started to try and give it a crack, but new ESPHome versions have removed custom climate components.

[image]

Have you gotten this working with newer ESPHome? Thank you!!

0xadam, 2025-03-25:
@Verpz i installed 2024.12.4 as that's the last version with external component support on my Ubuntu machine using `pipx install esphome==2024.12.4`

But can also add support to the yaml using an external component (haven't tried it though) as described at https://community.home-assistant.io/t/fujitsu-ac-heat-pump-integration-via-esphome-esp32/407610/300?u=0xadam

`external_components:
  - source:
      type: git
      url: https://github.com/robertklep/esphome-custom-component
    components: [ custom, custom_component ] `

Verpz, 2025-03-25:
@0xadam it works. You are a legend thank you so much. I used the legacy esphome addon.

unreality, 2025-03-26:
@0xadam would you be willing to create a pull request to merge your group control code in?

0xadam, 2025-03-26:
@unreality sure! I had a quick look just now and see that the repo I forked from had modified the base code in your repo (and also had some formatting changes) so it's not a simple cherry pick unfortunately. But will hopefully find the time in the next little bit.

0xadam, 2025-03-28:
@dymondj what are the odds after this sitting around for over a year?!

I'll have a look at your traces and repo to see what the differences between our implementations are. I never thought to look at what happened when outlets were enabled on the controller.

I'll see if I can get yours going on my esp32 too. Does it support single core? I haven't had issues with stability but have heard that repo is more stable, plus the full configuration through yaml is nice (I just didn't want to have to mess around working out the esphome framework and how to integrate to that nicely)

> @0xadam, looks like we were working on this in parallel :)
> 
> I've got some more data around the zone interactions now available at https://github.com/dymondj/fujitsu-airstage-h-dissector. These Wireshark traces have comments in them and I've also updated the dissector to break down each packet with the fields I could work out (zones enabled on controller, zone group assignments, active zones, etc).
> 
> I did do an initial cut of getting this working in https://github.com/dymondj/esphome-fujitsu-halcyon/tree/zones. Keen to see this capability get completed somewhere so we can all benefit. I've personally found this code base a little more stable for my setup than https://github.com/FujiHeatPump/esphome-fujitsu (at least for my use on ESP32).
> 
> Cheers!

0xadam, 2025-04-01:
@dymondj hey mate. Haven't had a chance to test yours out yet but did see there was some stuff commented out in esphome-fujitsu-halcyon.cpp that appeared to be to do with showing the zones only if enabled, and also had zones 1 and 2 always enabled for testing.

dymondj, 2025-04-03:
Hey @0xadam, yup sorry about that. Working with @Omniflux to see if there is a solution to get the zones enabled dynamically as part of the capabilities the indoor unit advertises on start up. That PR is still partly a work in progress so hopefully we can get that ironed out soon. I'll probably just move enabling these to the config file for now but initially I had permanently enabled zones 1 & 2 for testing on my system.

I've dual core ESP32s so I haven't been able to test yet on a single core. Please let me know how you go. Cheers.

0xadam, 2025-06-22:
I haven't updated esphome at all sorry.

Verpz, 2025-06-22:
@LukeScerri you have to pin your AC ESP version at the version above. You can use the legacy esphome addon to flash your device at that 2024 software. That's the last version to support custom components. All other devices you can continue to update just don't touch the AC one.

LukeScerri, 2025-06-27:
Thanks @Verpz and @0xadam .. I rolled everything back and it still wasn't working.. then OUT-OF-THE-BLUE it started working. You know how it is, I got side-tracked by work and life stuff. 
Something weird is going on though, i'm not convinced its totally stable.. I will keep the AC ESP pinned to the 2024 version moving forward. However, I have continuous warnings relating to a component blocking, and wonder if that is causing some dramas after some time.

[component:237]	Component <unknown> took a long time for an operation (332 ms).
[component:238]	Components should block for at most 30 ms.

0xadam, 2025-06-28:
@LukeScerri  Glad to hear you got it working again! Yep I can relate ha. Just FYI I get those component blocking messages too. I assume it's something to do with the low baud rate of the device (500) so comms takes a while. IIRC they were there before zone control support too (but maybe it didn't block for as long). I've had no issues with it working at all - it's been flawless the whole time. 

@dymondj Sorry I haven't had a chance to test your fork. I've not had time to get around to flashing it, as sadly I don't have much time for fiddling these days.

j-marz, 2025-11-08:
Building on @0xadam 's work and taking some shortcuts with Cursor, i have a fork of this library and have zone control working on an ESP32. See https://github.com/j-marz/FujiHeatPump

It supports a 2-zone system, but refers to them using my labels (upstairs and downstairs). The zone groups also use my labels (day, night, all).