unreality/FujiHeatPump #16: List of compatible LIN chips

daxliniere, 2024-09-05:
Hey @unreality,
I've been having trouble finding the MCP2025 IC at a reasonable price in the UK and it got me thinking that it would be great to have a list of all the known compatible ICs for this project. Mostly CANBUS ICs show up.

MCP2025
ATA663211
MCP2002A
MCP2002B

Do you know of any others?

All the best,
Dax.

dzungpv, 2024-10-18:
You can use TJA1027T

unreality, 2024-10-22:
Ive not tested anything except the MCP2025-330 and [MCP2003](https://ww1.microchip.com/downloads/en/DeviceDoc/20002230G.pdf) , but i'm reasonably confident any LIN transceiver will work. The only reason the MCP2025 was chosen to test with was that its a through-hole mount component that was available at the time.

alexjurkiewicz, 2024-11-26:
Has anyone tried the TJA1028T/3V3/20 ? It includes a voltage regulator like the MCP2025.

daxliniere, 2024-11-26:
No sorry, I haven't.

klim123123, 2025-06-11:
> Ive not tested anything except the MCP2025-330 and [MCP2003](https://ww1.microchip.com/downloads/en/DeviceDoc/20002230G.pdf) , but i'm reasonably confident any LIN transceiver will work. The only [...]

Could you please please provide schematic to connect MCP2003.

jpa73, 2025-11-07:
[image]

I used the MCP2021-500.  Note that the RXD output on pin 1 is 5v as it is the '500' part number.  The RX input on the ESP32 Serial is 3.3v so R1, and R2 form a voltage divider to drop the output to around 3.3v

jpa73, 2026-08-27:
> Hey [@unreality](https://github.com/unreality), I've been having trouble finding the MCP2025 IC at a reasonable price in the UK and it got me thinking that it would be great to have a list of all the known compatible ICs for this project. Mostly CANBUS ICs show up.
> 
> MCP2025 ATA663211 MCP2002A MCP2002B TJA1027T (no built-in voltage regulator
> 
> I have also found units on AliExpress that use TJA1020/TJA1021 for a few bucks. Any idea on their suitability? https://www.mouser.co.uk/new/nxp-semiconductors/nxp-tja1021-transceivers/
> 
> Do you know of any others?
> 
> All the best, Dax.

I did this with MCP2021.

https://github.com/unreality/FujiHeatPump/issues/16#issuecomment-3499956943

peterskeetero, 2026-08-27:
I have used a Microchip Technology MCP2021P-330E/SN,
https://au.mouser.com/en/ProductDetail/Microchip-Technology/MCP2021P-330E-SN?qs=hort8UWU32oHFTyz3SCDkQ%3D%3D&countryCode=AU&currencyCode=AUD
to talk to my Fujitsu ducted system..works a treat.