unreality/FujiHeatPump #18: Error 12.1 (Wired remote controller communication error)

robfox92, 2025-04-02:
Thanks for the great project.

I've run in to Error 12.1 (Wired remote controller communication error) earlier, and now  my secondary controller fails to bind to the bus. 

I've taken a trace, and I get the following:

```
<-- 0 81 0 6 1B A4 0 20   mSrc: 0 mDst: 1 mType: 0 write: 0 login: 0 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 A0 0 6 1B A4 C0 0   mSrc: 0 mDst: 32 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:32 acError:0 
<-- 0 A0 0 6 1B A4 0 20   mSrc: 0 mDst: 32 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 81 0 6 1B A4 0 20   mSrc: 0 mDst: 1 mType: 0 write: 0 login: 0 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 A0 0 6 1B A4 0 20   mSrc: 0 mDst: 32 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 81 0 6 1B A4 0 20   mSrc: 0 mDst: 1 mType: 0 write: 0 login: 0 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 A0 0 6 1B A4 0 20   mSrc: 0 mDst: 32 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 81 0 6 1B A4 0 20   mSrc: 0 mDst: 1 mType: 0 write: 0 login: 0 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 0 A0 0 6 1B A4 0 20   mSrc: 0 mDst: 32 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:10 cTemp:0 acError:0 
<-- 20 81 20 0 0 0 E 0   mSrc: 32 mDst: 1 mType: 2 write: 0 login: 0 unknown: 1 onOff: 0 temp: 0, mode: 0 cP:0 uM:0 cTemp:7 acError:0 
<-- 0 A0 20 1F F 4 1 0   mSrc: 0 mDst: 32 mType: 2 write: 0 login: 1 unknown: 1 onOff: 1 temp: 15, mode: 7 cP:1 uM:0 cTemp:0 acError:0
<-- 20 A1 0 6 1B 4 0 0   mSrc: 32 mDst: 33 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:0 cTemp:0 acError:0 
--> 21 81 0 6 1B 24 1 0   mSrc: 33 mDst: 1 mType: 0 write: 0 login: 0 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:1 uM:2 cTemp:0 acError:0 
<-- 20 81 0 6 1B 4 0 0   mSrc: 32 mDst: 1 mType: 0 write: 0 login: 0 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:0 uM:0 cTemp:0 acError:0 
<-- 0 A0 0 6 1B A4 1 20   mSrc: 0 mDst: 32 mType: 0 write: 0 login: 1 unknown: 1 onOff: 0 temp: 27, mode: 3 cP:1 uM:10 cTemp:0 acError:0 
```
Setup for this was to turn off the AC at the breaker, power my esp32 (running a modbus server I wrote on top of this project), then turn on the AC while debug logging was enabled.

This is the entire "beginning" of the trace, I can provide a longer one if required. I note that:

1. Primary sends login message to Unit
2. "Start" sends login response message to Primary
3. Primary sends status to Secondary with `login:1, unknown:1, cP:0, uM:0`
4. Secondary sends status to Unit with `login: 0, unknown:1, cP:1, uM:2`

I can see in [#2](https://github.com/unreality/FujiHeatPump/issues/2#issuecomment-1003244172) that the secondary should "forward that frame back to the unit" - should this be untouched? As it is, the secondary is changing some values. I'm wondering if it should just be forwarding the message to the unit if it's the first status it's received?

unreality, 2025-04-02:
Sending on the frame unaltered is worth looking at - in my setup the initial frame was not modified so the flow of the initial login frame went unchanged along this path:

Unit --> Primary --> Secondary --> Unit

Also ensure that the initial login frame is forwarded to the Secondary controller. If the Primary sends it back to the Unit, the secondary will not be able to join. Once the communication 'loop' is established it cant be changed without a power cycle afaik.

Other things to note are the OEM remotes will store their last-good settings and fail to re-bind unless the DIP switches are toggled.

robfox92, 2025-04-03:
Thanks for the quick response!

I've tried toggling both DIP switches on the back (with the circuit breaker disconnected) and I get the same login flow.

Just to be clear, when you're talking about the initial login frame, it's `mType:2`? Or are you referring to the first status message?

robfox92, 2025-04-06:
Would anyone be able to provide a debug trace of "normal" power up/function?

robfox92, 2025-04-20:
Turns out I had a hardware issue - I think I burnt out my bus transceiver. Running with a new breadboard works fine. For anyone wanting a debug trace of the "normal" flow (I have forked this repo in part to get "prettier" debug printing:

```
-- 00 A0 00 02 14 44 00 20 @ 23      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP0,uM 4,cTmp 0,err0
-- 00 81 00 02 14 A4 00 20 @ 24      START ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 81 00 02 14 A4 00 20 @ 25      START ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 A0 00 02 14 A4 00 20 @ 25      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 81 00 02 14 A4 00 20 @ 26      START ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 A0 00 02 14 A4 00 20 @ 26      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 81 00 02 14 A4 00 20 @ 26      START ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 A0 00 02 14 A4 00 20 @ 27      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 81 00 02 14 A4 00 20 @ 27      START ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 00 A0 00 02 14 A4 00 20 @ 28      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP0,uM10,cTmp 0,err0
-- 20 81 20 00 00 00 0E 00 @ 28    PRIMARY ==>      UNIT:   LOGIN,wr0,log0,unk1,on0,tmp 0,mod0,cP0,uM 0,cTmp 7,err0
-- 00 A0 20 1F 0F 04 01 00 @ 28      START ==>   PRIMARY:   LOGIN,wr0,log1,unk1,on1,tmp15,mod7,cP1,uM 0,cTmp 0,err0
>> 20 A1 00 02 14 04 00 00 @ 29    PRIMARY ==> SECONDARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP0,uM 0,cTmp 0,err0
<< 21 81 00 02 14 24 01 00 @ 29  SECONDARY ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP1,uM 2,cTmp 0,err0
-- 00 A0 00 02 14 A4 03 20 @ 29      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM10,cTmp 1,err0
>> 20 A1 00 02 14 04 33 00 @ 29    PRIMARY ==> SECONDARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
<< 21 81 00 02 14 04 33 00 @ 29  SECONDARY ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
-- 00 A0 00 02 14 A4 03 20 @ 30      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM10,cTmp 1,err0
>> 20 A1 00 02 14 04 33 00 @ 30    PRIMARY ==> SECONDARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
<< 21 81 00 02 14 04 33 00 @ 30  SECONDARY ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
-- 00 A0 00 02 14 A4 03 20 @ 31      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM10,cTmp 1,err0
>> 20 A1 00 02 14 04 33 00 @ 31    PRIMARY ==> SECONDARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
<< 21 81 00 02 14 04 33 00 @ 31  SECONDARY ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
-- 00 A0 00 02 14 A4 03 20 @ 32      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM10,cTmp 1,err0
>> 20 A1 00 02 14 04 33 00 @ 32    PRIMARY ==> SECONDARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
<< 21 81 00 02 14 04 33 00 @ 32  SECONDARY ==>      UNIT:  STATUS,wr0,log0,unk1,on0,tmp20,mod1,cP1,uM 0,cTmp25,err0
-- 00 A0 00 02 14 A4 03 20 @ 33      START ==>   PRIMARY:  STATUS,wr0,log1,unk1,on0,tmp20,mod1,cP1,uM10,cTmp 1,err0
```

The specific Fujitsu hardware in my case is:
- Indoor unit: ARTG36LHTA
- Controller: UTY-RVNYN

closed 2025-04-20 by robfox92